Carleton-SRL/SPOT #8: Thrust Estimation

TheElectricDream, 2021-09-20:
Currently, it is assumed that each thruster can provide a maximum thrust of 0.25 N. This is inaccurate, because the amount of thrust out of each thruster is also a function of which thrusters are being used.

Sydneymoses, 2023-10-16:
Controls sub team is working on this

c-bash, 2024-02-12:
See Lugini and Romano's ["A ballistic-pendulum test stand to characterize small cold-gas
thruster nozzles"](https://www.sciencedirect.com/science/article/pii/S0094576508003603) for reference.

TheElectricDream, 2024-09-30:
I have written code to accomplish this task for RED and it seems to be pretty decent - but it requires knowing the CG, mass, and inertia of the platforms at the time the data is collected. I will write an entry about it soon.