Carleton-SRL/SPOT #17: E-Stop Does not Stop Arm

TheElectricDream, 2021-09-30:
Hitting the e-stop does not stop the robotic manipulator. This could be potential catastrophic for students who are not aware of the issue.

c-bash, 2022-06-30:
Was in the laboratory today. E-stop did indeed stop the arm. This issue may be resolved.

TheElectricDream, 2023-03-04:
Great, I will close this issue!

TheElectricDream, 2023-03-12:
Yikes, yeah this should not be closed. The e-stop does not stop the arm. Maybe it's inconsistent (which is also scary), but this needs to be fixed before any inexperienced student does arm experiments. It caught be off guard and I had to unplug the arm manually:

https://www.dropbox.com/s/ocs539ub5l527cw/HCVR_Overhead_main_20230312121350_20230312121409.avi?dl=0

c-bash, 2023-03-12:
When we re-do the wiring on RED we can make sure that the power to the arm also goes through the emergency MOSFET.

c-bash, 2023-12-05:
I don't think I've been able to fix it yet, so best not to close this issue since we might have a repeat of previous events. The problem seemed to be that the arm still had a 'small' amount of current running through it even when the e-stop was pressed. Whether the current was coming from a residual amount in the 12V line, or whether it is somehow getting power from elsewhere (e.g., through the Xavier) remains unknown.