Carleton-SRL/SPOT #28: Error in GUI v3_07

ProfSteveUlrich, 2022-03-14:
When clicking "Confirm Options" in the GUI, the following error is reported:
-------
Unrecognized method, property, or field 'cla' for class 'matlab.ui.control.UIAxes'.

Error in GUI_v3_07/AddPlatform (line 126)
            app.ExperimentPreview.cla;

Error in GUI_v3_07/ConfirmOptionsButtonPushed (line 431)
                AddPlatform(app,evalin("base","platformSelection"));
 
Error using matlab.ui.control.internal.controller.ComponentController/executeUserCallback (line 386)
Error while evaluating Button PrivateButtonPushedFcn.
-------

TheElectricDream, 2022-03-14:
This is a known problem when using MATLAB versions newer then 2020a. Due to Mathwork's apparently inability to maintain consistent functions in their GUI API, the GUI is usually restricted to a couple versions of MATLAB at one time. Future updates may try to improve compatibility, or at the very least place a warning to users attempting to use newer versions. For now, please use 2020a and confirm it is functional. I will leave this issue open for now for other users.

TheElectricDream, 2022-06-14:
Closing this issue as it's a MATLAB issue.

closed 2022-06-14 by TheElectricDream