Carleton-SRL/SPOT #29: Stop Experiment should write 0 to thrust

JeffHough, 2022-03-17:
After E-stopping hitting and "Stop Experiment", if the E-stop is released, the thrusters will fire with their previous write value. The platforms then have to be shut off, and then turned back on again. It would be best if hitting "Stop Experiment" also writes the thrust to 0.

Kirkados, 2022-03-17:
As an interim fix, clicking "Stop Experiment" a second time should write the thrusters to zero. However, I agree, this should be done more reliably with the first click of "Stop Experiment"

TheElectricDream, 2022-03-17:
It's supposed to do exactly this by running a compiled executable called CleanShutdown.elf. Are you working with the latest GitHub version? We noticed it was not working last week but it started working when I used newer code... unconfirmed but I suspect some versions of code out there don't run it. Please check your working GUI code and confirm that the StopExperimentButton looks like this:

[image]

TheElectricDream, 2022-06-14:
Seems to work in experiment now. Please ensure you have the latest software version.

closed 2022-06-14 by TheElectricDream