Carleton-SRL/SPOT #50: E-stop doesn't cut 12 V line when GPIO pins are connected

c-bash, 2023-08-08:
The 12V line doesn’t get cut when the GPIO pins are connected to the thruster board. I believe the e-stop drain is getting connected to GND through the Jetson GPIO pins.

c-bash, 2023-08-09:
Seems to be solved with the GPIO pulldown resistor going to GND instead of the MOSFET drain. BLACK and BLUE have a ~0V potential when e-stop is activated. RED has a floating 0.2-0.4V potential, which is likely a faulty hardware connection somewhere. Should look at the thruster breadboard connections: Are any MOSFETS dead/dying? Is there a short due to wires being too long or being too close? Etc.

TheElectricDream, 2026-06-10:
With the new PCB design and the fact that we no longer use the GPIO pins on the Xavier, we can now close this issue! The e-stop does indeed cut all thrusters.