Carleton-SRL/SPOT #52: Upgrade Drain Tubes on Air Compressor

c-bash, 2023-09-15:
Air compressor: Current drain tubes for points B and C are cracking and quite short with liquid currently being collected by a plastic water bottle. 

An upgrade is required such that the drain tubes are replaced and, ideally, not pointing directly at the platforms. A more semi-permanent collection basin should be at the end of the tubes.

Kirkados, 2023-09-15:
Perhaps an elbow could run drain tubes down towards the floor where an absorbent material (like a dog pee pad?) absorbs the ejected moisture.

TheElectricDream, 2023-09-18:
A pro tip as well, I've stopped using the bottle and I use a bundle of shop towels to catch the liquid. Just less gross that way.

c-bash, 2024-01-12:
This has been completed! In future, we likely want to move the drain box to a better location since currently it blocks the bottom right drawer. 

[image]

closed 2024-01-12 by c-bash