Carleton-SRL/SPOT #53: Fix the warning message when clicking on "Refresh Platform Software"

TheElectricDream, 2023-09-20:
The warning message headers and message still refer to the Raspberry Pi, so this needs to be updated for the next software release.

[image]

TheElectricDream, 2023-09-24:
- Remove a leftover termination block I found in the "Check Connections" subsystem.
- Update version numbers to 4.1 and modify the warning message on startup to check for R2023a and R2023b.
- Update the default SPOT GUI mat file to correct some mistakes and ensure compatibility with 4.1.

TheElectricDream, 2023-09-24:
- As indicated by Steve, it would be great if we could make the connection LED update in real-time to notify users if there is a drop in the connection status. This could be done by having a function ping the platforms every minute or so. This change would have to be done in a way that does not freeze the rest of the GUI.

TheElectricDream, 2023-09-24:
- Try a heuristic-based approach for the control mixer to improve results.
- Investigate alternative controllers (or implement best capstone controller) to find new default controller. Should be simple enough for future researchers but not so complex as to cause confusion.

TheElectricDream, 2023-12-03:
- In the real-time viz, it should show the current time for the experiment somewhere.
- There should be a button to refresh the list of available Simulink diagrams so that you don't need to restart the GUI when you make a new diagram.

c-bash, 2024-01-12:
- Move BLACK's magnet gate control wire from 3.3V to a GPIO on the Xavier NX. Add functionality in the GUI to toggle this GPIO pin so that we can detach the magnets without turning off power to BLACK.

TheElectricDream, 2024-03-13:
Fix the incorrect labels in the data logging subsection of the simulink template.

TheElectricDream, 2024-09-30:
This list is now closed with the release of SPOT 4.1 Pre-Release.1, available for download here: https://github.com/Carleton-SRCL/SPOT/releases/tag/4.1.0-prerelease.1-hotfix.1

closed 2024-09-30 by TheElectricDream