Carleton-SRL/SPOT #56: SPOT GUI Indicates (0,0,0) for Platforms Positions, but LEDs Stay On

TheElectricDream, 2023-11-25:
I believe the issue is caused by the PhaseSpace program that sends the data crashing in the background. This would result in the LEDs being on (PhaseSpace initialized), but there is no UDP to send the data, and the GUI is stuck in a pseudo-functional state. I can add a check in the GUI to record the ID of the PhaseSpace executable when it is started, and then I can check that it is still running every couple of seconds. If it crashed, I can toggle the GUI PhaseSpace switch to OFF to properly shut down the PhaseSpace cameras and then I can provide a warning to the user to simply "try again".

TheElectricDream, 2023-11-25:
The current workaround is to restart the ground station computer.

c-bash, 2024-04-01:
Another potential fix is to replace the StreamData.exe file in your current project to the most up-to-date version [found on the wiki](https://github.com/Carleton-SRCL/SPOT/blob/main/Template_Files/Resources/StreamData.exe)

TheElectricDream, 2024-09-22:
Resolved in [4.1](https://github.com/Carleton-SRCL/SPOT/releases/tag/4.1.0-prerelease.1)

closed 2024-09-22 by TheElectricDream