Carleton-SRL/SPOT #61: Multiple errors when ending an experiment in the SPOT GUI

natashadawnn, 2024-01-15:
Running experiments today and noticed that multiple errors popped up in the command window when the "Stop Experiment" button was pressed. It does not seem to affect anything, but looks like there is a bug.

[image]

Logs:
~~~
Note: For the complete log, run the following command in the MATLAB command window:
system(hwobj,'cat /home/spot-blue/ExperimentSoftware/MATLAB_ws/R2023a/C/Users/SPOT/Documents/SPOT-4.0.0-RC.4_-_APF/SPOT-4.0.0-RC.4/SPOT-4.0.0-RC.4/Projects/APF_Path_Planning_Revised/APF_Path_Planning_2023a.log')
Checking connection to RED platform...
Checking connection to BLACK platform...
Checking connection to BLUE platform...
SUCCESS: The process "StreamData.exe" with PID 18576 has been terminated.
Dot indexing is not supported for variables of this type.

Error in GUI_v4_0_Main/StartSimulationButtonPushed (line 2012)
            dataPacket = simOut.dataPacket;

Error in appdesigner.internal.service.AppManagementService/executeCallback (line 172)
                callback(appOrUserComponent, event);

Error in matlab.apps.AppBase>@(source,event)executeCallback(ams,app,callback,requiresEventData,event) (line 62)
            newCallback = @(source, event)executeCallback(ams, ...
 
Error using matlab.ui.control.internal.controller.ComponentController/executeUserCallback
Error while evaluating Button PrivateButtonPushedFcn.
~~~

Will be running more experiments tomorrow so I can add more detail to this ticket if necessary.

TheElectricDream, 2024-01-15:
That looks like someone tried to save simulation data (hence the dataPacket = simOut.dataPacket), but no simulation data exists. 

Please document the exact steps needed to replicate this error and I will investigate.

TheElectricDream, 2024-01-17:
After some digging around, the problem was resolved by copying the contents from the problematic Simulink diagram to a fresh copy of the template diagram from RC.4. It appears the problem is isolated to this one copy, and the most likely cause is that the Simulink diagram was tweaked in MATLAB 2023b and then ported to 2023a.

I am closing this issue. If the problem reappears, I will reopen this issue.

closed 2024-01-17 by TheElectricDream