Carleton-SRL/SPOT #64: Add Wiki Entry Regarding Mac OS Compatibility

TheElectricDream, 2024-03-04:
Title is self-explanatory

adamvigneron, 2024-03-04:
First, the easy one -- Mac OS paths use the forward slash, Windows paths use the backward slash.

Affected files:
``Setup_Environment.m``
``GUI_v4_0_Main.mlapp``

One possible fix is to replace all path-related uses of backslash with the ``filesep`` command:
[filesep in MATLAB](https://www.mathworks.com/help/matlab/ref/filesep.html)

Also possible to build paths using the ``fullfile`` command:
[fullfile in MATLAB](https://www.mathworks.com/help/matlab/ref/fullfile.html)

...according to this thread, we could just YOLO and only use forward slashes:
[forward slash thread](https://www.mathworks.com/matlabcentral/answers/117110-dealing-with-and-windows-vs-unix-for-path-definition)

adamvigneron, 2024-03-04:
Second, as the NVIDIA add-on isn't currently supported for Mac OS, experiments are not possible AND simulations need a few more things commented out:

``GUI_v4_0_Main.mlapp``, lines 1973 and 1999
the project ``*.slx`` file, blocks corresponding to experiment logic and the NVIDIA interface.

A detailed list will be maintained on the wiki.

adamvigneron, 2024-03-05:
@TheElectricDream , let's see how we can do this in an efficient manner.

Per the comments above, I've made changes to three files:
``SetupEnvironment.m``
``GUI_v4_0_Main.mlapp``
``BetaMacs.slx`` (my project copy of the template)

I've pushed the changes to my SPOT fork:
https://github.com/adamvigneron/AdamVigneron-SPOT/commit/2eeefe4f9d93ccc0c31c05fe6068f345f108d513

where it's possible to see the plaintext M-file changes but not possible to see the binary changes.  Luckily, the binary files can be compared to their ``Template_Files/``equivalents using MATLAB functionality for File Compare.

How do you want to go about capturing this in the wiki?
How do you want to go about capturing this in the codebase?

TheElectricDream, 2024-09-30:
I have create a Wiki entry on this subject here: https://github.com/Carleton-SRCL/SPOT/wiki/Regarding-MacOS-Compatibility

However, we will not be creating and supporting a MacOS compatible diagram - I do not have a MacOS system available and the lab stance is that it should not be used with SPOT. I've entered the steps as you laid out in the Wiki entry, but these steps may change with future versions and I will not dedicate time to keeping those steps up to date. Users who are simply MUST use MacOS should expect to be on their own and will need to sort out compatibility on a case-by-case basis.

closed 2024-09-30 by TheElectricDream

adamvigneron, 2024-09-30:
The wiki entry looks great, thanks Alex!

As discussed, there are two additional code snippets we could add to GUI_v4_0_Main.mlapp:

```
if ismac
    fprintf('GUI_v4_0_Main.StartSimulationButtonPushed:\n');
    fprintf('  MATLAB Coder for NVIDIA unavailable on Mac OS;\n');
    fprintf('  pre-simulation switch of HardwareBoard disabled.\n\n')
else
    set_param(activeConfigObj,'HardwareBoard','None');
end
```

```
if ismac
    fprintf('GUI_v4_0_Main.StartSimulationButtonPushed:\n');
    fprintf('  MATLAB Coder for NVIDIA unavailable on Mac OS;\n');
    fprintf('  post-simulation switch of HardwareBoard disabled.\n\n')
else
    set_param(activeConfigObj,'HardwareBoard','NVIDIA Jetson');
end
```