Carleton-SRL/SPOT #66: Control Panel Status Light is Intermittent

c-bash, 2024-03-19:
Seen mainly for BLACK so far, the blue status light on the control panel flickers in and out randomly. When this happens during an experiment, the thrusters turn off which messes up the control of BLACK for ~5 seconds.

When the LED goes out (not necessarily flashing red, but just is no longer blue), the RX signal being picked up by the receiver is not "good" or perhaps is lost for a few seconds. 

Areas to check in the circuit: 
(1) the RX antenna on BLACK and related solder connections
(2) the receiver on BLACK and related solder connections
(3) the Arduino Pro Mini on BLACK and related solder connections

Kirkados, 2024-03-19:
[Please disregard this comment if things have changed since the initial design]

During the initial design, when the RX signal is not received by the Arduino Mini for 5 seconds, it automatically triggers the e-stop and slowly flashes the status light off -> red -> off -> red -> etc. 

This was a fail-safe mechanism, so the platform automatically stops when e-stop communication is lost for a sufficiently long time.

A suggested action is to move RX antenna on Black—perhaps it is intermittently not receiving the e-stop TX signal in its current position.

c-bash, 2024-03-19:
Hi Kirk, we are still using the initial design for the Arduino Pro Mini code, so still very relevant.

I think your suggestion on moving the antenna to a new location is a good and easy solution to check first. Thanks!