Carleton-SRL/SPOT #71: Thruster PCB Breadboard

c-bash, 2024-09-09:
An open task from last year that was not accomplished. Needs to replace the solderless breadboard on each platform with a (more permanent) PCB. Design needs to be reviewed by the admin team before any submissions for manufacturing.

c-bash, 2024-10-12:
This is the breadboard to be replaced on each platform: 

[image]

The circuit is shown in part here:

[image]

The MOSFETs should not be soldered directly to the board since sometimes these components get damaged and must be replaced. Instead, they should be connected through a female header connector (such as https://www.digikey.ca/en/products/detail/sullins-connector-solutions/PPTC031LFBN-RC/810143, note, this specific connector has not be confirmed to be compatible with the dimensions of the MOSFET). 

Most components (e.g., MOSFETS, some connectors, resistors) have been ordered by last year's capstone team. These components can be found in the WIRING drawer of the lab.

sean-charlton, 2024-11-18:
The current design of the PCB uses 0.4mm traces on a single side. 
[image]

sean-charlton, 2024-11-18:
Schematic of the thruster circuit
[image]

adamvigneron, 2025-06-17:
Met with Alison Fong, a hardware technologist in the Department of Electronics.

For the next version of the PCB, Alison agrees that:
- the harness should include an explicit ground connection.  She agreed that the circular connector currently used on the breadboard is a good candidate.
- a Schottky diode connected across the solenoid terminals would protect the solenoids against inductive kickback (a so-called “flyback diode”).
- an opto-coupler could be used to fully isolate the 5V Xavier from the 24V thruster board.

Alison suggested that all future PCBs avoid the use of 90º angles in their traces.

Alison agreed to check the KiCAD design of the updated PCB.